teemuatlut/Marlin #10: Configuration TMC 2208 unclear

NSchnitzler, 2017-09-08:
I have changed the configuration_adv.h to: 
`#define HAVE_TMC2208

#if HAS_TRINAMIC // TMC2130 & TMC2208 & TRAMS
.....
  #define X_IS_TMC2208
  //#define X2_IS_TMC2208
  #define Y_IS_TMC2208
  //#define Y2_IS_TMC2208
  #define Z_IS_TMC2208
  //#define Z2_IS_TMC2208
  #define E0_IS_TMC2208`

but the compiler answers with:
`exit status 1
'class TMC2208Stepper' has no member named 'isEnabled'
`

teemuatlut, 2017-09-08:
Update the TMC2208Stepper library to the latest version from github.

teemuatlut, 2017-09-08:
Yes, and that is why I advised to update from github.
The Arduino IDE library manager versions lag behind either because I haven't been confident enough in a version to publish it yet or simply because I haven't yet gotten around to it.

NSchnitzler, 2017-09-08:
ok compiling completes
thank you

NSchnitzler, 2017-09-08:
ok communication works, 
perhaps an improvement idea: how about to combine the upper micro step settings 
#define MICROSTEP_MODES {16,16,16,16,16}
with the TMC settings:
  #define X_CURRENT         600  // rms current in mA. Multiply by 1.41 for peak current.
  #define X_MICROSTEPS        16  // 0..256
  #define Y_CURRENT         600
  #define Y_MICROSTEPS        16

so that you have to define them at one single point instead of keeping them synched.

teemuatlut, 2017-09-08:
I kinda like that all the TMC related configurations are within the same section. But I'll think about it.

NSchnitzler, 2017-09-08:
I know what you mean, encapsulation is a good thing. Or ignore the upper values if tmc is enabled for this axis.

closed 2017-10-18 by teemuatlut