teemuatlut/Marlin #9: Marlin-Trams

detoxity, 2017-09-08:
Hello, you can upload new trams-marlin? In this branch when i click home x (in repetier host) my axis z come down. and printer does not respond to movement commands.
Or maybe you will give some advice on how to configure the firmware before compiling.

teemuatlut, 2017-09-08:
I've pushed a few commits.
Currently TRAMS for is configured to use stallGuard homing.

If you Z came down (a bit?) when you started homing X, your Z direction is likely incorrect. Whenever you start the homing process on Marlin, it first raises the nozzle to avoid scraping the heated bed.

First before homing, try moving an axis and see if the right axis responds and moves in the correct direction.
I'd also advise to completely power cycle every time you encounter weird behavior. The driver registers aren't completely initialized on start up.

detoxity, 2017-09-09:
I achieved normal movement along the axes. but after some time the printer hangs up and stops responding, the house function for x and y does not work.

detoxity, 2017-09-09:
Does your printer print with this firmware? Maybe I'm not configuring it correctly?

teemuatlut, 2017-09-09:
Yes, the hangs are a problem that I have not fixed yet.

I don't yet print with it because I don't yet have a printer to put it in. Working on that part though.

detoxity, 2017-10-04:
Hello, you have not solved the problem with the firmware for TRAMS? Or maybe you know what kind of working firmware for TRAMS.

teemuatlut, 2017-10-04:
I'm tied to other projects currently and still lacking printers.

The [Trinamic Marlin fork](https://github.com/trinamic/TRAMS-Firmware) should be in working condition but is over a year old.