teemuatlut/Marlin #11: [FR] a couple of new ideas for the trinamic drivers

dot-bob, 2017-10-01:
With the recent announcement that the Prusa I3 MK3 will be using the TMC2130's it will be interesting to see the firmware when it is finally released to see what they are doing with the TMC2130 drivers.

One feature that seems interesting is their skipped step detection.  Upon a skipped step they appear to lift the head a few mm and re-home X and Y then resume printing.

Another feature I have been thinking of is the ability to slow the print speed and drop stepper current if the drivers are getting too hot.

On the flip-side perhaps we could detect quick accelerations in the planner and temporarily bump the current.  The prusa bumps up the current to the extruder stepper when changing filaments on their MMU kit all be it through gcode in their transition script and not firmware.

teemuatlut, 2017-10-01:
Without knowing too much about their implementation, my best guess is that when the driver detects a high load, it'll pull a diag pin high. The change is then detected by the firmware that stops the queue, raises head, homes XY and resumes printing.

I've already got a form of driver temperature monitoring. If a driver has triggered the OverTemperature Pre-Warn, the firmware will slightly decrease current until the condition is cleared. Wouldn't be too hard to also decrease the print speed as well.
For increasing current, I'm utilizing the TMC automatic current scaling when stealthChop is being used. This should match the stepper current even when higher step rates are used.
When not using stealthChop, we could make use of coolStep, but that hasn't been implemented yet. This basically does what you ask as it monitors the motor load and adjusts current appropriately.

For quick movements you can also use the hybrid mode where the driver will switch to spreadCycle.

dot-bob, 2017-10-05:
Looking through the data sheet... another feature that may be worthwhile to implement is the double edge step.  This would be beneficial to 8 bit microcontrollers as it effectively halves the required step rate.

teemuatlut, 2017-10-05:
Perhaps.
However the same effect can be had with using a smaller microstepping setting. And with that the steps are produced more evenly. With interpolation the smoothness should still be there as the motor still sees 256 steps. Personally I'm using only 1/8 microstepping on my Z axis.

closed 2019-06-17 by teemuatlut