teemuatlut/Marlin #12: unexpected movements during prints

NSchnitzler, 2017-10-02:
the printer moves extruding outside of the object or crosses the inside area, the movements made were not visible in the GCODE.

[image]
[image]
[image]

teemuatlut, 2017-10-02:
You're rocking the TMC2208 steppers, right? Those only know to step when there is a step pulse and they will not make movements on their own.
I suspect the issue is outside of the drivers themselves. Are the extra movements always on the same position/place in the print? Can you test with other type of drivers? What branch are you using?

NSchnitzler, 2017-10-03:
Yes right, i am using the 2208. I also think that this is a firmware problem, yesterday i started a large print and in the middle of the layer the printer stops with `Current build time: 60.92 minutes
WARNING: Firmware unresponsive.  Attempting to force continue...` after two minutes he prints a line and then it cycles between the error and a little bit of printing. 
Currently i am using your dev branch. When will the TMC2208 drivers appear in a stable release? Or which branch should i use at the moment?

teemuatlut, 2017-10-03:
Use the ```master``` branch as that has the most recent developments.
Upstream support will come after most of the 2.0 release hassle has been cleared. Marlin is in a semi-feature-freeze currently.

NSchnitzler, 2017-10-04:
I am shure that i am using the master branch of this repo. I will try reflashing and reducing the serial baud rate

NSchnitzler, 2017-10-05:
OK reflashing and reducing the Baudrate seems to work for the moment. 
Thank you

closed 2017-10-05 by NSchnitzler