teemuatlut/Marlin #13: compile problems

NSchnitzler, 2017-10-29:
Hi teemuatlut,

i had to switch to a new PC and used the chance to upgrade all libs and so on. 
I took your TMC stepper lib and the actual version of Arduino, but now i am getting this compile error:
`
sketch\Marlin_main.cpp: In instantiation of 'void tmc_status(TMC&, TMC_AxisEnum, TMC_debug_enum, float) [with TMC = TMC2208Stepper]':

sketch\Marlin_main.cpp:9953:73:   required from here

Marlin_main.cpp:9892: error: 'class TMC2208Stepper' has no member named 'isEnabled'

         case TMC_ENABLED: serialprintPGM(st.isEnabled() ? PSTR("true") : PSTR("false")); break;

                                                         ^

Marlin_main.cpp:9914: error: no matching function for call to 'TMC2208Stepper::TPWMTHRS()'

             uint32_t tpwmthrs_val = st.TPWMTHRS();`

teemuatlut, 2017-10-29:
Try using the TMC2208 library version straight from github. I haven't pushed an update in a while for the Arduino Library Manager.

NSchnitzler, 2017-10-29:
I Could remember this information and replaced the installed ALM lib with your actual one. But it didn't help.

NSchnitzler, 2017-10-29:
OK really strange behavior:
with windows (which always worked) i get the error mentioned above
with mac (which never compiled successfully before) works now!

so it works for me now, that's fine, even if i don't understand why/why not.

thank you for your assistance

teemuatlut, 2017-10-29:
Double check that you're using the github version of the library. If it compiles on mac, then it shouldn't be a problem with the code. I'll try to push a new version for the Library Manager in the coming week that might solve your problems.

closed 2018-01-07 by teemuatlut