teemuatlut/Marlin #14: PR into Marlin

psavva, 2017-11-02:
### Feature Request

Hi teemuatlut,. any hope in getting TMC driver support in the 2.0 Marlin?

What is current supported at the moment in the official 2.0 branch?
Which TMC drivers do you recommend I purchase with the idea that it will be supported in Marlin 2.0 either now or in the near future?

teemuatlut, 2017-11-02:
Definitely there's hope. I just need to get to it.
But at first it may be just AVR and DUE (and Teensy likely) because the driver libraries depend on a couple of other standard libraries and I don't know if those are available for Re-Arm or STM32 or ESP32 etc.

I believe in the 2.0 it's on par with 1.1.6, but both are quite a bit behind from my dev branch.

TMC2130 has the best feature set and is my primary focus.

psavva, 2017-11-02:
Ordered 6 of these puppies:)
http://www.watterott.com/en/SilentStepStick-TMC2130

In am currently testing the RE-ARM board with the rest of the Marlin team.
Let me know if there is some way I can test the drivers on the Re-Arm, and will be happy to get feedback to you.

teemuatlut, 2017-11-08:
The Re-Arm does not have a proper SPI library and the HAL uses internal SPI commands for everything. So the Re-Arm is not compatible with the TMC2130Stepper library. For now at least.

psavva, 2017-11-08:
I see plenty of libraries on the web for SPI driver LPC chipset... 

What can I do to make it a reality to get these drivers working on the Re-Arm?

teemuatlut, 2017-11-08:
Perhaps but they are not part of Marlin but they might work if you install it. As long as the include works and they follow the same interface as the libraries for AVR/Due/Teensy.

teemuatlut, 2017-11-08:
The best I think would be if the SPI implementation from the Re-Arm HAL would get moved to a library format. Even if it was local to the Marlin project.

psavva, 2017-11-08:
I have created https://github.com/MarlinFirmware/Marlin/issues/8334

closed 2018-01-07 by teemuatlut