teemuatlut/Marlin #15: Skipped step detection with TMC2130 drivers

teemuatlut, 2017-12-25:
@thinkyhead Hey I'm opening up a ticket here to have a quick discussion about how to implement this and I reckon you know the gcode handler/buffer better than anyone.

The skipped step detection has been requested a few times and was hinted towards on one Tom's videos. Prusa Research even has this implemented already.

So in order for us (me-too mentality) to have a similar feature for mainstream Marlin, I was thinking that the best way would be to utilize the endstops using interrupts feature. Basically if the interrupt is triggered when we are NOT homing an axis, then we inject a `G28 XY` command into the buffer as the very next command to execute.
I haven't yet looked at how to code this, so would you have any thoughts or suggestions regarding the implementation method?

thinkyhead, 2018-02-05:
Hey, just found this as I was going through issues. They tend to pile up!

This feature definitely makes sense when using sensorless homing. It should be something we can turn on and off. It would replace the current behavior of the abort-on-endstop-hit feature with something a bit simpler, which is to re-home and resume instead of completely aborting the print.

Currently when we want to interrupt the G-code processing we do it by having a G-code handler go into a loop. For this it might make sense to add a pause flag to the G-code parser so it can stop processing G-code altogether while doing the re-home. That would allow the re-home to occur more quickly than if a `G28 X Y` was queued. The move back to the print position would be automatically done by the next G1 move processed, but only for Absolute (`G90`) mode. For a print job entirely done in relative mode it would be a little trickier.

kubik256, 2018-02-15:
Sounds promising, may be Prusa's MK3 FW can be good starting point...
https://github.com/prusa3d/Prusa-Firmware/tree/MK3