teemuatlut/Marlin #16: [FR] DIAG1 stallGuard as additional filament runout sensor

q3ok, 2018-02-21:
While the DIAG1 on X and Y axis can be used for sensorless homing, there I found additional usage of the stallGuard for E axis. It can be connected (together, or without) to filament sensor pin, to trigger the M600 when the filament is clogged, and detected by TMC2130.
Together with filament sensor and connected with OR logic gate made from two diodes and resistors that can detect any problems with the filament.
I stringly believe it's not a big problem, as it should be just enabling the DIAG1 as stallGuard on E stepstick, and there will be more with the cable connections.

teemuatlut, 2018-02-21:
Sounds like a good idea.
However, I'd rather not go the way of having users making OR gates for themselves. Merely checking two pins should be easier from end user perspective.

q3ok, 2018-02-21:
I was thinking about logic gate, as soon there will be no enough free pins to connect something more :) 5 stepsticks + software SPI to avoid problems with SD cards, its 8 pins, so the whole AUX2 on ramps will be used, two pins for filament runout sensor and stallguard from, and only 4 pins are left, when the RRD fan extension will be used - even no more easily accesible pins. More pins free will be available only if not every endstop will be used, but using the sensorless homing on X and Y can give 2 free pins from MAX endstops.