teemuatlut/Marlin #17: [FR] StealthChop/Hybrid/SpreadCycle on-line change

q3ok, 2018-02-23:
Currently the mode in which the TMC operates muse be set in configuration. There should be a option to change it, even while printing - eg. printing during the day can be run in SpreadCycle, when going to sleep moving to StealthChop or Hybrid.
Additionally to limit step losses that should have also a option to automatically reduce acceleration and/or max speeds while changing to stealthchop, and again while back to spreadcycle unleashing the power.

teemuatlut, 2018-02-23:
I might look into it after 1.1.9 is released and I can get back to adding more features.

q3ok, 2018-02-23:
https://github.com/q3ok/Marlin/commit/f1eb1ff662b45646331de6642fbf61d5cfe649b3
 I just wanted to make it work (tested & working), but it's not implemented in a good manner. Based on Marlin 1.1.x branch.

closed 2019-01-21 by teemuatlut