teemuatlut/Marlin #18: Typo causes compilation error

lienbacher, 2018-03-17:
Hi all!
Just updating my printers drivers to TMC2208, followed instructions and was welcomed with a compilation error in [L#385/#386](https://github.com/teemuatlut/Marlin/blob/fc3bc0b2d31a0cfb7de5a6a0826e55bb4485dee9/Marlin/stepper_indirection.cpp#L385).
Code says:
`    st.hysterisis_start(3);
    st.hysterisis_end(2);`

Changing hysterisis to hysteresis fixed the error:
'    st.hysteresis_start(3); 
    st.hysteresis_end(2);'

Seems this typo was [fixed](https://github.com/teemuatlut/TMC2208Stepper/commit/583d6dbe41c0276b37b0f3af15b1846925302461) just recently in the TMC2208 library and now breaks the marlin implementation.

teemuatlut, 2018-03-17:
This is because a recent bugfix required changes to both Marlin and the libraries.
You need to either downgrade the library or upgrade Marlin to bugfix-1.1.x branch.

lienbacher, 2018-03-20:
Gotcha, Thank you!

closed 2018-03-20 by lienbacher