teemuatlut/Marlin #19: Trouble getting UART connection to TMC2208

lienbacher, 2018-03-20:
Hello! 

I'm having problems troubleshooting my uart connection to my driver. So far I have only installed it on the Y-axis. When calling M122 I receive a 0x00 register. I have taken issue #9396 into account and made sure pins are ok. I am using an Arduino Mega with Ramps 1.4 and the latest TMC2208 from Watterott on only one Axis till I get it right. with the 3 pads for the uart connection, and soldered a 1K smd resistor between one pad to the center pad, and solderbridged the other pad to the center pad, TX is connected to the pin with the resistor. I tripple checked there is a 1k resistor to TX with the multimeter and connected to the correct pin, even measured it from chip to end of cable to rule out any bad connections in between. I have also tried different pin-configurations by editing pins_RAMPS.h referencing issue #9396, still receiving 0x00. 
Everything is configured as described [here](http://marlinfw.org/docs/features/tmc_drivers.html), HAS_TMC2208 is enabled, and Y_IS_TMC2208 is also defined.

I am still a n00b, so I guess I am overlooking something. I would be very thankful if somebody could share some troubleshooting tips as I am running out of ideas. Guess next step would be to update from release to the developer branch ...

Thank you very much in advance!

teemuatlut, 2018-03-20:
Perhaps some pictures would clear your situation. I know the board design has changed since I've received mine.

teemuatlut, 2018-03-21:
CLK needs to be connected to ground to enable the internal clock. See the PDF you linked.

lienbacher, 2018-03-21:
Thank you! Now that you said it I see it, plus the J1 solder bridge that disables the clock pin anyway.

Still no luck :( Used a different Arduino Mega and a fresh 2208 board. So I guess it is not a hardware fault. Also double checked if the connection was on the correct pins. 

This is how I configured the resistor on the pads. The Pin that has the 1K resistor on it connects to Tx, Rx is connected to an Interrupt pin as described int he issue above.

Any ideas from here? Simple troubleshooting process I can walk from here? I even checked if my ramps board was defined correctly, so I guess it should use pins_RAMPS.h ...
[image]

thank you!

teemuatlut, 2018-03-21:
Try to get the library example sketches working first. Possibly even without the RAMPS board.
Also make sure you have 12V (Vm) and 5V (Vio) lines powered to the driver before trying to communicate with it.

lienbacher, 2018-03-21:
I found the issue and as I expected it was something rather n00bish. I was using v0.0.4 of the 2208 library with 1.1.8 release of marlin, where I manually fixed the typo with histerisis/histeresis but did not check for other changes. I have now updated the 2208 library to v0.1.1 and used the bugfix 1.1.x branch of marlin now. 

I am getting a valid register now! Yay! Only one thing is weird, I am receiving data for X2 instead of Y. I do not even know if this is a real issue, but it leaves me wondering, since X2 is not defined, I only defined Y_IS_TMC2208 ... you got any intel on that? 

thank you very much for your fantastic work!

teemuatlut, 2018-03-21:
Good to hear you've got it working.

The axis codes should be defined by what you've selected but there have been some changes that could cause something like that. It's likely just the axis label however and you should just try if the driver(s) work.

lienbacher, 2018-03-21:
The driver works fine. It even worked without the proper uart connection. My y axis moves just fine! It only seems the label, can you hint me to a file/line in the code where I can check?

teemuatlut, 2018-03-21:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/tmc_util.cpp#L217

The likely cause is the different order between:
`enum TMC_AxisEnum : char { TMC_X, TMC_Y, TMC_Z, TMC_X2, TMC_Y2, TMC_Z2, TMC_E0, TMC_E1, TMC_E2, TMC_E3, TMC_E4 };`
and
`const static char* const tmc_axes[] PROGMEM = { ext_X, ext_X2, ext_Y, ext_Y2, ext_Z, ext_Z2, ext_E0, ext_E1, ext_E2, ext_E3, ext_E4 };`

lienbacher, 2018-03-21:
confirmed, thank you! I've changed the enum to `enum TMC_AxisEnum : char { TMC_X, TMC_X2, TMC_Y, TMC_Y2, TMC_Z, TMC_Z2, TMC_E0, TMC_E1, TMC_E2, TMC_E3, TMC_E4 };` and it is reporting Y. Previously it was reporting X2 and it was also reporting X2 in the stst line, where it now reports X. I honestly have no clue what stst means, so I cannot tell if this is wrong or correct.

teemuatlut, 2018-03-21:
`stst` is short for standstill.

teemuatlut, 2018-03-21:
`x` should be interpreted as a check mark, meaning `true`.

lienbacher, 2018-03-21:
gotcha. Thank you very much for your help!

closed 2018-03-21 by lienbacher