teemuatlut/Marlin #23: Roxy at Slack

teemuatlut, 2018-05-11:
@Roxy-3D Can you still be reached in Slack? I wanted to talk to you about a few UBL related things.

Roxy-3D, 2018-05-11:
I got an invite and joined...   But I guess I never went back.  Mostly because it never told me anything was happening there.

What's up?

teemuatlut, 2018-05-11:
Here's what I wrote earlier:
> Hey, could we remove the `M421` output from `M503`? It takes a really long time to print out and can even pause a print during that time. Also it tends to clutter the serial output when I'm just trying to see my acceleration settings.

I've of course done this for myself but maybe there would be others who might be as frustrated with it as I was.

---

For a few days now, I've been tuning my UBL mesh on a new frame and I noticed that `G26` doesn't seem to have travel specific feedrates. I think it at least doubles the procedure time when the head was moving side to side across the 300mm bed plate. Perhaps this could be something to look at? Or maybe I'm missing some setting.

---

Then there's something that actually managed to damage my bed surface.
[image]
I'd like to believe my mesh is fine, as indicated my the printed horizontal lines near the edge. But as soon as the head is right on the maximum axis length, it dipped a few millimeters down. My `UBL_Z_RAISE_WHEN_OFF_MESH` is set at 3.6 which should clear the bed just fine. I'll need to reflash my board as I'm not 100% if the value was changed before or after the latest upload.
I seem to recall that there was an issue related to this some time ago and I believe it was addressed. Have you seen this behavior yourself?

Roxy-3D, 2018-05-11:
I think I agree we should remove the M421 lines.   Especially with a 10x10 or 15x15 mesh, it takes way too long.    I think the intent was to have all the information to recreate the settings if you move to a different version of the firmware available.   But it is too much.   Perhaps we could put an entire row (10 or 15 values) on a line and that would be 'tolerable'.

I see what you are talking about.   The mesh values are correct...   But some how the connecting lines are too low on the edge.   It is very possible this just got fixed:  https://github.com/MarlinFirmware/Marlin/issues/10686

teemuatlut, 2018-05-11:
I understand the intent and I too would like to be access the output when required. Maybe just for safekeeping in a gcode script. The downside of the Due platform is that the flash / EEPROM emulation gets wiped when the board is reflashed.

My branch was based on 439e0cdd0f8b0701e5468111c37d5f2036e2d0d6 from yesterday, so it seems that particular merge was not included.

Roxy-3D, 2018-05-11:
How about we see where the discussion goes:   https://github.com/MarlinFirmware/Marlin/issues/10707

closed 2019-01-21 by teemuatlut