teemuatlut/Marlin #27: Compile error on AZTEEG G5- interrupt pins

monterman, 2018-10-22:
I get this compile error.   attached are

[Marlin.zip](https://github.com/teemuatlut/Marlin/files/2501667/Marlin.zip)

 my config file are attached.  please advise and help.  

Compiling .pioenvs\LPC1768\src\src\module\endstops.o
In file included from Marlin\src\module\endstops.cpp:36:0:
Marlin\src\module\../HAL/HAL_LPC1768/endstop_interrupts.h:54:8: error: #error "X_MIN_PIN is not an INTERRUPT capable pin."
#error "X_MIN_PIN is not an INTERRUPT capable pin."
^~~~~

Marlin\src\module\../HAL/HAL_LPC1768/endstop_interrupts.h:66:8: error: #error "Y_MIN_PIN is not an INTERRUPT capable pin."
#error "Y_MIN_PIN is not an INTERRUPT capable pin."
^~~~~
Marlin\src\module\../HAL/HAL_LPC1768/endstop_interrupts.h:78:8: error: #error "Z_MIN_PIN is not an INTERRUPT capable pin."
#error "Z_MIN_PIN is not an INTERRUPT capable pin."
^~~~~
Marlin\src\module\configuration_store.cpp: In static member function 'static void MarlinSettings::reset(int8_t)':
Marlin\src\module\configuration_store.cpp:1812:26: error: 'DEFAULT_ACCELERATION' was not declared in this scope
planner.acceleration = DEFAULT_ACCELERATION;
^~~~~~~~~~~~~~~~~~~~
Marlin\src\module\configuration_store.cpp:1812:26: note: suggested alternative: 'DEFAULT_MAX_ACCELERATION'
planner.acceleration = DEFAULT_ACCELERATION;
^~~~~~~~~~~~~~~~~~~~
DEFAULT_MAX_ACCELERATION
*** [.pioenvs\LPC1768\src\src\module\configuration_store.o] Error 1
*** [.pioenvs\LPC1768\src\src\module\endstops.o] Error 1
 [ERROR] Took 283.60 seconds
Environment LPC1768                     [ERROR]
 [ERROR] Took 283.61 seconds

kAdonis, 2018-10-24:
This is not specific for this Marlin Fork
Endstop Interrupts feature is not compatible to standard endstop pins on a Azteeg X5 GT
see PR #11202
you can try using different pins for your endstops or just disable Endstop Interrupts

default accelaration should be enabled
you have in Configuration.h:

`//#define DEFAULT_ACCELERATION          3000    // X, Y, Z and E acceleration for printing moves`

monterman, 2018-10-25:
@kAdonis thanks! your advise worked.

However I now have a new error: see below.  I don't knwo where to configure Chopper timing or if I need to disable.  I do not remember activating any stealth chop in my settings....

`Compiling .pioenvs\LPC1768\src\src\module\tool_change.o
Marlin\src\module\stepper_indirection.cpp: In function 'void tmc_init(TMCMarlin<TMC2660Stepper, AXIS_LETTER, DRIVER_ID>&, uint16_t, uint16_t, uint32_t, fl
oat, bool)':
Marlin\src\module\stepper_indirection.cpp:563:24: error: 'CHOPPER_TIMING' was not declared in this scope
int8_t timings[] = CHOPPER_TIMING; // Default 4, -2, 1

^~~~~~~~~~~~~~
*** [.pioenvs\LPC1768\src\src\module\stepper_indirection.o] Error 1
 [ERROR] Took 684.16 seconds
Environment LPC1768                     [ERROR]`

kAdonis, 2018-10-25:
You also disabled in your Configuration_adv.h:
`  //#define CHOPPER_TIMING { 4, -2, 1}`

carefully check your configs with a diff-tool against the default configs

you are probably wrong in this experimental fork of marlin
your hardware is supported in main marlin https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

monterman, 2018-10-28:
I got it to complie,  you were right, my conf advance file did not have the //#define CHOPPER_TIMING { 4, -2, 1}
However, the MARLIN https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x  does not have it either.
The only fork that actually has the Chopper Timing in the  conf_adv it is the Teemuatlut MASTER https://github.com/teemuatlut/Marlin

Thanks for your help!

Sincerely,

Andres