teemuatlut/Marlin #30: TMC2209 branch compile error

BioGeekJoey, 2019-06-17:
### Description

While trying to compile, I get an error caused by a missing argument. I did try to look into the problem but I have no experience with C/C++.  I tried looking for obvious errors but can't find anything.

#### Additional Information

`In file included from Marlin\src\module/stepper_indirection.h:60,
from Marlin\src\module/stepper.h:221,
from Marlin\src\Marlin.cpp:37:
Marlin\src\module/../feature/tmc_util.h: In member function 'int16_t TMCMarlin<TMC2209Stepper, AXIS_LETTER, DRIVER_ID, AXIS_ID>::homing_threshold()':
Marlin\src\module/../feature/tmc_util.h:241:39: error: no matching function for call to 'TMCMarlin<TMC2209Stepper, AXIS_LETTER, DRIVER_ID, AXIS_ID>::SGTHRS()'
return TMC2209Stepper::SGTHRS();
^
In file included from Marlin\src\module/stepper_indirection.h:59,
from Marlin\src\module/stepper.h:221,
from Marlin\src\Marlin.cpp:37:
.piolibdeps\TMCStepper_ID5513\src/TMCStepper.h:994:8: note: candidate: 'void TMC2209Stepper::SGTHRS(uint8_t)'
void SGTHRS(uint8_t B);
^~~~~~
.piolibdeps\TMCStepper_ID5513\src/TMCStepper.h:994:8: note:   candidate expects 1 argument, 0 provided
Compiling .pioenvs\LPC1769\src\src\feature\babystep.cpp.o
Compiling .pioenvs\LPC1769\src\src\feature\backlash.cpp.o
*** [.pioenvs\LPC1769\src\src\Marlin.cpp.o] Error 1
In file included from Marlin\src\feature\../module/stepper_indirection.h:60,
from Marlin\src\feature\../module/stepper.h:221,
from Marlin\src\feature\babystep.cpp:30:
Marlin\src\feature\../module/../feature/tmc_util.h: In member function 'int16_t TMCMarlin<TMC2209Stepper, AXIS_LETTER, DRIVER_ID, AXIS_ID>::homing_threshold()':
Marlin\src\feature\../module/../feature/tmc_util.h:241:39: error: no matching function for call to 'TMCMarlin<TMC2209Stepper, AXIS_LETTER, DRIVER_ID, AXIS_ID>::SGTHRS()'
return TMC2209Stepper::SGTHRS();
^
In file included from Marlin\src\feature\../module/stepper_indirection.h:59,
from Marlin\src\feature\../module/stepper.h:221,
from Marlin\src\feature\babystep.cpp:30:
.piolibdeps\TMCStepper_ID5513\src/TMCStepper.h:994:8: note: candidate: 'void TMC2209Stepper::SGTHRS(uint8_t)'
void SGTHRS(uint8_t B);
^~~~~~
.piolibdeps\TMCStepper_ID5513\src/TMCStepper.h:994:8: note:   candidate expects 1 argument, 0 provided
*** [.pioenvs\LPC1769\src\src\feature\babystep.cpp.o] Error 1`

oechslein, 2019-06-17:
Problem is in tmc_util.h:
```
    #if USE_SENSORLESS
      inline int16_t homing_threshold() { return TMC2209Stepper::SGTHRS(); }
```
And TMC2209Stepper class does only define/implement a write method but no read method:
```
		// W: SGTHRS
		void SGTHRS(uint8_t B);
```

teemuatlut, 2019-06-17:
Added now https://github.com/teemuatlut/TMCStepper/commit/bd5cba6a0bf5f439746a295e5acd2a515570cddb

Sensorless homing might be supported (don't quite remember all the conditional macros by heart) but it very likely will not work yet and I won't have the time to test it for a while.

BioGeekJoey, 2019-06-17:
Great! I will it tonight! Will report back.

BioGeekJoey, 2019-06-17:
Edits did fix it, next issue arose haha. I think it is complaining about menu items now in regard to sensorless homing. Undefining Sensorless homing did allow the compiler to complete this time! Uploading it to an SKR 1.3 to test soon.

```cpp
In file included from Marlin\src\lcd\menu\menu_tmc.cpp:31:
Marlin\src\lcd\menu\menu_tmc.cpp: In function 'void menu_tmc_homing_thrs()':
Marlin\src\lcd\menu\menu.h:304:56: error: no matching function for call to 'TMenuItem<MenuItemInfo_int8>::action_edit(const char [2], int16_t*, const uint8_t&,
const uint8_t&, void (&)())'
MenuItem_##TYPE ::action ## VARIANT(__VA_ARGS__); \
^
Marlin\src\lcd\menu\menu.h:340:62: note: in expansion of macro '_MENU_ITEM_VARIANT_P'
#define MENU_ITEM_EDIT_CALLBACK(TYPE, LABEL, ...)            _MENU_ITEM_VARIANT_P(TYPE, _edit, false, PSTR(LABEL), PSTR(LABEL), ## __VA_ARGS__)
^~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_tmc.cpp:215:35: note: in expansion of macro 'MENU_ITEM_EDIT_CALLBACK'
#define TMC_EDIT_STORED_SGT(ST) MENU_ITEM_EDIT_CALLBACK(int8, MSG_##ST, &stepper##ST.stored.homing_thrs, stepper##ST.sgt_min, stepper##ST.sgt_max, refresh_homing_thrs_##ST);
^~~~~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_tmc.cpp:231:7: note: in expansion of macro 'TMC_EDIT_STORED_SGT'
TMC_EDIT_STORED_SGT(X);
^~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:191:17: note: candidate: 'static void TMenuItem<NAME>::action_edit(const char*, TMenuItem<NAME>::type_t*, TMenuItem<NAME>::type_t, TMenuItem<NAME>::type_t, screenFunc_t, bool) [with NAME = MenuItemInfo_int8; TMenuItem<NAME>::type_t = signed char; screenFunc_t = void (*)()]'
static void action_edit(PGM_P const pstr, type_t * const ptr, const type_t minValue, const type_t maxValue, const screenFunc_t callback=nullptr, const bool live=false) {
^~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:191:17: note:   no known conversion for argument 2 from 'int16_t*' {aka 'short int*'} to 'TMenuItem<MenuItemInfo_int8>::type_t*' {aka 'signed char*'}
Marlin\src\lcd\menu\menu_tmc.cpp:215:75: error: cannot convert 'int16_t*' {aka 'short int*'} to 'MenuItemInfo_int8::type_t*' {aka 'signed char*'}
#define TMC_EDIT_STORED_SGT(ST) MENU_ITEM_EDIT_CALLBACK(int8, MSG_##ST, &stepper##ST.stored.homing_thrs, stepper##ST.sgt_min, stepper##ST.sgt_max, refresh_homing_thrs_##ST);
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:308:99: note: in definition of macro '_MENU_ITEM_VARIANT_P'
draw_menu_item ## VARIANT ## _ ## TYPE(encoderLine == _thisItemNr, _lcdLineNr, PLABEL, ## __VA_ARGS__); \
^~~~~~~~~~~
Marlin\src\lcd\menu\menu_tmc.cpp:215:35: note: in expansion of macro 'MENU_ITEM_EDIT_CALLBACK'
#define TMC_EDIT_STORED_SGT(ST) MENU_ITEM_EDIT_CALLBACK(int8, MSG_##ST, &stepper##ST.stored.homing_thrs, stepper##ST.sgt_min, stepper##ST.sgt_max, refresh_homing_thrs_##ST);
^~~~~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_tmc.cpp:231:7: note: in expansion of macro 'TMC_EDIT_STORED_SGT'
TMC_EDIT_STORED_SGT(X);
^~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:108:134: note:   initializing argument 5 of 'void draw_menu_item_edit_int8(bool, uint8_t, const char*, const char*, MenuItemInfo_int8::type_t*, ...)'
FORCE_INLINE void draw_menu_item_edit_##NAME (const bool sel, const uint8_t row, PGM_P const pstr, PGM_P const pstr2, TYPE * const data, ...) { \
^
Marlin\src\lcd\menu\menu.h:117:50: note: in expansion of macro '_DEFINE_DRAW_MENU_ITEM_SETTING_EDIT'
#define DEFINE_DRAW_MENU_ITEM_SETTING_EDIT(NAME) _DEFINE_DRAW_MENU_ITEM_SETTING_EDIT(MenuItemInfo_##NAME::type_t, NAME, MenuItemInfo_##NAME::strfunc)
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:122:1: note: in expansion of macro 'DEFINE_DRAW_MENU_ITEM_SETTING_EDIT'
DEFINE_DRAW_MENU_ITEM_SETTING_EDIT(int8);             // 123, -12   right-justified
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:304:56: error: no matching function for call to 'TMenuItem<MenuItemInfo_int8>::action_edit(const char [2], int16_t*, const uint8_t&,
const uint8_t&, void (&)())'
MenuItem_##TYPE ::action ## VARIANT(__VA_ARGS__); \
^
Marlin\src\lcd\menu\menu.h:340:62: note: in expansion of macro '_MENU_ITEM_VARIANT_P'
#define MENU_ITEM_EDIT_CALLBACK(TYPE, LABEL, ...)            _MENU_ITEM_VARIANT_P(TYPE, _edit, false, PSTR(LABEL), PSTR(LABEL), ## __VA_ARGS__)
^~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_tmc.cpp:215:35: note: in expansion of macro 'MENU_ITEM_EDIT_CALLBACK'
#define TMC_EDIT_STORED_SGT(ST) MENU_ITEM_EDIT_CALLBACK(int8, MSG_##ST, &stepper##ST.stored.homing_thrs, stepper##ST.sgt_min, stepper##ST.sgt_max, refresh_homing_thrs_##ST);
^~~~~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_tmc.cpp:234:7: note: in expansion of macro 'TMC_EDIT_STORED_SGT'
TMC_EDIT_STORED_SGT(Y);
^~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:191:17: note: candidate: 'static void TMenuItem<NAME>::action_edit(const char*, TMenuItem<NAME>::type_t*, TMenuItem<NAME>::type_t, TMenuItem<NAME>::type_t, screenFunc_t, bool) [with NAME = MenuItemInfo_int8; TMenuItem<NAME>::type_t = signed char; screenFunc_t = void (*)()]'
static void action_edit(PGM_P const pstr, type_t * const ptr, const type_t minValue, const type_t maxValue, const screenFunc_t callback=nullptr, const bool live=false) {
^~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:191:17: note:   no known conversion for argument 2 from 'int16_t*' {aka 'short int*'} to 'TMenuItem<MenuItemInfo_int8>::type_t*' {aka 'signed char*'}
[... cut ...]
Marlin\src\lcd\menu\menu.h:117:50: note: in expansion of macro '_DEFINE_DRAW_MENU_ITEM_SETTING_EDIT'
#define DEFINE_DRAW_MENU_ITEM_SETTING_EDIT(NAME) _DEFINE_DRAW_MENU_ITEM_SETTING_EDIT(MenuItemInfo_##NAME::type_t, NAME, MenuItemInfo_##NAME::strfunc)
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:122:1: note: in expansion of macro 'DEFINE_DRAW_MENU_ITEM_SETTING_EDIT'
DEFINE_DRAW_MENU_ITEM_SETTING_EDIT(int8);             // 123, -12   right-justified
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pioenvs\LPC1769\src\src\lcd\menu\menu_tmc.cpp.o] Error 1
```

teemuatlut, 2019-06-18:
https://github.com/teemuatlut/Marlin/commit/816455b974d99a2b6543acc4553e17a4ef794213#diff-4b4767cc241fbed4c80cc9961091d592

Your build log doesn't match with the code base. Try updating.

oechslein, 2019-06-18:
I confirm that with the latest TMCStepper lib it compiles even when enabling SENSORLESS_HOMING.
Once I got my 2209 I will also test.

BioGeekJoey, 2019-06-20:
Enabling sensorless homing seems to cause all steppers to not get enabled. M211 will display they are not enabled even after sending a G0/1 seconds ago.

Disabling seemed to be what fixed it. Had to leave so can provide any more info atm. I copied my config as a new branch on my marlin repo.

[bd98d03ee4fae9d4dac45636a23829c44e35353e]

teemuatlut, 2019-06-20:
I need debugging information to do anything about it.
Enabled checks two things; EN pin is low and off time is not set to zero.

BioGeekJoey, 2019-06-20:
@teemuatlut anything in particular that would help the most to diagnose this?

teemuatlut, 2019-06-21:
EN pin can be tested with a multimeter and `M43`.
Off time can be read with `M122`.

It also helps to know if the drivers at any point lock up.

BioGeekJoey, 2019-06-21:
Alright. I'm currently generating my UBL mesh and will do some test prints today en tomorrow. I won't have time to poke around with my multimeter until Sunday at best.

Will report back!

BioGeekJoey, 2019-07-01:
Took my longer than I had hoped to get back to this. 

On the current 2.0.x branch (c28e08c8495a0cfa4b382632e8d853079d47b8c0) the steppers do work even with sensorless_homing enabled. It doesn't home sensorless, but that might be due to my hardware config. I may need to short some pins.

Guess I can close this?

closed 2019-07-16 by BioGeekJoey