teemuatlut/Marlin #31: BigTreeTech SKR V1.3 Dual Z Stepper don't functional

ShadowD71, 2019-09-29:
I have a BigTreetech SKR V1.3. This is equipped with five Stepperdriver ... 4xTMC2130 (SPI) and 1xTMC2208 (UART).

Extract from config_adv.h:

 #define Z_DUAL_STEPPER_DRIVERS
#if ENABLED (Z_DUAL_STEPPER_DRIVERS)
  // #define Z_DUAL_ENDSTOPS
  #if ENABLED (Z_DUAL_ENDSTOPS)
    #define Z2_USE_ENDSTOP _XMAX_
    #define Z_DUAL_ENDSTOPS_ADJUSTMENT 0
  #endif
#endif

Now, if I press Move Z on the display, then ONLY the Z Stepper will move. If I press Move X, the X Stepper and the Y Stepper will move. The cables of the stepper are all plugged in correctly. The second Z-engine is plugged in E1. The drivers are also configured correctly. X, Y, Z, E2 = TMC2130 and E0 = TMC2208.

What am I doing wrong ??

teemuatlut, 2019-09-29:
Start with this
http://marlinfw.org/docs/hardware/tmc_drivers.html#troubleshooting

teemuatlut, 2019-09-29:
The possible reasons why your motor isn't turning are
* The driver doesn't get configured properly -> Bad connection / invalid configuration
  * TMC drivers are enabled when `off time != 0` and enable is `low`
* The EN pin doesn't get pulled low
* The STEP pin isn't toggled when moving

ShadowD71, 2019-09-30:
It works now. Since it was not documented in Marlin, I mistakenly assumed that when setting #define Z_DUAL_STEPPER_DRIVERS the driver of E1 will be used automatically. Unfortunately, you have to use the Z2 driver there. From then on it went immediately.

closed 2019-09-30 by ShadowD71